rufengsuixing/luci-app-adguardhome #140: 关于openwrt编译luci-app-adguardhome，执行文件路径错误的问题

sss132919, 2023-01-15:
lean最新源码编译openwrt x86 加入最新的luci-app-adguardhome的插件，执行文件路径显示/usr/bin/AdGuardHome , 但是这个路径在ssh里查看了是个文件夹，导致无法启动，实测将路径修改为/usr/bin/AdGuardHome/AdGuardHome之后不论是更新还是启动都恢复正常，希望可以在源码端处理一下，不然每次都要手动修改感觉比较麻烦，特别是保留配置升级的情况下

lsq726, 2023-01-29:
/usr/bin/AdGuardHome/AdGuardHome
👍

>

lsq726, 2023-03-27:
> 可以

请教还做了什么操作么我修改了后还是无法启动

lsq726, 2023-03-27:
> lean最新源码编译openwrt x86 加入最新的luci-app-adguardhome的插件，执行文件路径显示/usr/bin/AdGuardHome , 但是这个路径在ssh里查看了是个文件夹，导致无法启动，实测将路径修改为/usr/bin/AdGuardHome/AdGuardHome之后不论是更新还是启动都恢复正常，希望可以在源码端处理一下，不然每次都要手动修改感觉比较麻烦，特别是保留配置升级的情况下

请教还做了什么操作么我修改了后还是无法启动

lsq726, 2023-03-28:
> > > lean最新源码编译openwrt x86 加入最新的luci-app-adguardhome的插件，执行文件路径显示/usr/bin/AdGuardHome , 但是这个路径在ssh里查看了是个文件夹，导致无法启动，实测将路径修改为/usr/bin/AdGuardHome/AdGuardHome之后不论是更新还是启动都恢复正常，希望可以在源码端处理一下，不然每次都要手动修改感觉比较麻烦，特别是保留配置升级的情况下
> > 
> > 
> > 请教还做了什么操作么我修改了后还是无法启动
> 
> 做什么操作都没用。https://github.com/liuran001/openwrt-packages.git 用这位老哥的luci，编译之前把源码自带的包删掉，network里也不要选中AdGuardHome，让luci脚本自己下载

请教具体怎么删除自带的包？  network里是menuconfig中的？