rufengsuixing/luci-app-adguardhome #142: Does not work correctly in the latest OpenWRT

fizsef, 2023-10-30:
Is the project no longer supported? Does not work correctly in OpenWRT version 23.05, AGH version 107.40. Please update and fix.

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version
- [ ] I checked to make sure that this issue has not already been filed

### Problem Description

### Proposed Solution

### Alternatives Considered

### Additional Information

jinxiao, 2024-01-28:
I don't think this repo is still under maintainence. There is a simple way to install AGH now. Stop all your AGH instance, use opkg to install adguardhome then start via /erc/init.d script.

fizsef, 2024-01-29:
> I don't think this repo is still under maintainence. There is a simple way to install AGH now. Stop all your AGH instance, use opkg to install adguardhome then start via /erc/init.d script.

Yes, there is an easy installation method. But at the same time, it is impossible to manage from Luci (stop the service, restart, edit, etc.). The point of this project was that everything can be done from Luci.

jinxiao, 2024-02-07:
> Yes, there is an easy installation method. But at the same time, it is impossible to manage from Luci (stop the service, restart, edit, etc.). The point of this project was that everything can be [...]

but the problem is, this plugin is doing too much more than simply manage the status of ADG, and make things too complicated than it is.