rufengsuixing/luci-app-adguardhome #145: fw4 重定向53 不工作

liyiranlab, 2024-01-02:
### 提交之前

在你提交问题之前请回答以下问题
**你可以删掉提交之前这个部分**

- [ ] 我用着最新版的 luci-app-adguardhome
- [ ] 我检查了没有人提交同样的问题

### 问题详细信息

* **AdGuard Home luci的版本:**
  * 
* **AdGuard Home server的版本:**
  * 
* **操作系统和版本:**
  * 

### 希望的执行结果

### 实际的执行结果

### 日志(重要)

### 截图

<details><summary>截图:</summary>

</details>

### 更多的信息

xmapst, 2024-04-12:
```sh
添加 转发到adguardhome 1053端口
nft add table inet adguardhome
nft add chain inet adguardhome prerouting "{ type nat hook prerouting priority -110; policy accept; }"
nft add rule inet adguardhome prerouting "meta nfproto { ipv4, ipv6 } udp dport 53 counter redirect to :1053 comment \"ADGUARDHOME HIJACK\""

# 清理
! nft --check list table inet adguardhome > "/dev/null" 2>&1 || nft delete table inet adguardhome
```

iseku, 2024-04-16:
可能是没有安装iptables-nft和iptables-mod-nat-extra，安装上就兼容fw4了

fxvbv, 2024-05-01:
有没有办法关闭默认的ipt防火墙，只使用nft防火墙，每次一启动这个插件，防火墙一直在报使用旧版协议。

xmapst, 2024-05-05:
> 有没有办法关闭默认的ipt防火墙，只使用nft防火墙，每次一启动这个插件，防火墙一直在报使用旧版协议。

将iptables规则用nft重写即可

S0MEW0LF, 2024-07-22:
> ```shell
> 添加 转发到adguardhome 1053端口
> nft add table inet adguardhome
> nft add chain inet adguardhome prerouting "{ type nat hook prerouting priority -110; policy accept; }"
> nft add rule inet adguardhome prerouting "meta nfproto { ipv4, ipv6 } udp dport 53 counter redirect to :1053 comment \"ADGUARDHOME HIJACK\""
> 
> # 清理
> ! nft --check list table inet adguardhome > "/dev/null" 2>&1 || nft delete table inet adguardhome
> ```

ssh上输入那三行命令后有效，但是路由器重启后就失效了。怎么可以让它一直生效呢？

gdanmac, 2024-08-31:
> ssh上输入那三行命令后有效，但是路由器重启后就失效了。怎么可以让它一直生效呢？

openwrt firewall4（FW4）自定义防火墙规则的方法

GoogleBeEvil, 2024-09-01:
https://github.com/kongfl888/luci-app-adguardhome/compare/master...starllll:luci-app-adguardhome:master
有人改写了  但是我编译出来仍然是失效的

link-fgfgui, 2024-09-02:
> ssh上输入那三行命令后有效，但是路由器重启后就失效了。怎么可以让它一直生效呢？

可以在openwrt启动项>本地启动脚本里加入这三句话

MIAOBUSI, 2024-11-20:
https://github.com/MIAOBUSI/luci-app-adguardhome，我自己弄了个修复版本，可以试试，我自己用着没问题了

9dark9, 2024-12-03:
> [https://github.com/MIAOBUSI/luci-app-adguardhome，我自己弄了个修复版本，可以试试，我自己用着没问题了](https://github.com/MIAOBUSI/luci-app-adguardhome%EF%BC%8C%E6%88%91%E8%87%AA%E5%B7%B1%E5%BC%84%E4%BA%86%E4%B8%AA%E4%BF%AE%E5%A4%8D%E7%89%88%E6%9C%AC%EF%BC%8C%E5%8F%AF%E4%BB%A5%E8%AF%95%E8%AF%95%EF%BC%8C%E6%88%91%E8%87%AA%E5%B7%B1%E7%94%A8%E7%9D%80%E6%B2%A1%E9%97%AE%E9%A2%98%E4%BA%86)

有没有编译好的，给个试试？

MIAOBUSI, 2024-12-05:
> 有没有编译好的，给个试试？

https://github.com/MIAOBUSI/luci-app-adguardhome/releases/tag/V1.9-1，已经上传

GoogleBeEvil, 2024-12-06:
> [https://github.com/MIAOBUSI/luci-app-adguardhome/releases/tag/V1.9-1，已经上传](https://github.com/MIAOBUSI/luci-app-adguardhome/releases/tag/V1.9-1%EF%BC%8C%E5%B7%B2%E7%BB%8F%E4%B8%8A%E4%BC%A0)

测试  转发53端口到adguard 模式是无效的。此模式下   无法过滤广告，面板里查询日志为0

MIAOBUSI, 2024-12-06:
> 测试 转发53端口到adguard 模式是无效的。此模式下 无法过滤广告，面板里查询日志为0

[image]
很奇怪。。我也是同一个版本，这边测试是可以的，你好固件确定是纯fw4么，没有混合fw3在里面吧

GoogleBeEvil, 2024-12-07:
> > > > > [https://github.com/MIAOBUSI/luci-app-adguardhome，我自己弄了个修复版本，可以试试，我自己用着没问题了](https://github.com/MIAOBUSI/luci-app-adguardhome%EF%BC%8C%E6%88%91%E8%87%AA%E5%B7%B1%E5%BC%84%E4%BA%86%E4%B8%AA%E4%BF%AE%E5%A4%8D%E7%89%88%E6%9C%AC%EF%BC%8C%E5%8F%AF%E4%BB%A5%E8%AF%95%E8%AF%95%EF%BC%8C%E6%88%91%E8%87%AA%E5%B7%B1%E7%94%A8%E7%9D%80%E6%B2%A1%E9%97%AE%E9%A2%98%E4%BA%86)
> > > > 
> > > > 
> > > > 有没有编译好的，给个试试？
> > > 
> > > 
> > > [https://github.com/MIAOBUSI/luci-app-adguardhome/releases/tag/V1.9-1，已经上传](https://github.com/MIAOBUSI/luci-app-adguardhome/releases/tag/V1.9-1%EF%BC%8C%E5%B7%B2%E7%BB%8F%E4%B8%8A%E4%BC%A0)
> > 
> > 
> > 测试 转发53端口到adguard 模式是无效的。此模式下 无法过滤广告，面板里查询日志为0
> 
> ![a724359f-d81e-4126-9245-8ff086330283](https://private-user-images.githubusercontent.com/20589467/393231869-ba4fbeb7-3ab8-4434-b731-69c6cef7a54a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM1MzIxNzMsIm5iZiI6MTczMzUzMTg3MywicGF0aCI6Ii8yMDU4OTQ2Ny8zOTMyMzE4NjktYmE0ZmJlYjctM2FiOC00NDM0LWI3MzEtNjljNmNlZjdhNTRhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjA3VDAwMzc1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBlNDY3MzMwNzk5ZGIwMjFlMTMzYTFmYTY1MGRkOGU0MTI5M2Q5YjRlZTkwNjBmZWNmZTBlYzc5Yjg1ZTFkNmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.lpMvgLFLh_k2wOquo_CzsSxQ6ZIPBiJQ2ZhtA6eigB8) 很奇怪。。我也是同一个版本，这边测试是可以的，你好固件确定是纯fw4么，没有混合fw3在里面吧

应该是春fw4的。在系统防火墙里看状态  没提示有iptables规则（如果有的话会提示）。固件是25.0.4  ，已安装包里也只有firewall4。

MIAOBUSI, 2024-12-07:
> > > > > > https://github.com/MIAOBUSI/luci-app-adguardhome，我自己弄了个修复版本，可以试试，我自己用着没问题了
> > > > > 
> > > > > 
> > > > > 有没有编译好的，给个试试？
> > > > 
> > > > 
> > > > [https://github.com/MIAOBUSI/luci-app-adguardhome/releases/tag/V1.9-1](https://github.com/MIAOBUSI/luci-app-adguardhome/releases/tag/V1.9-1%EF%BC%8C%E5%B7%B2%E7%BB%8F%E4%B8%8A%E4%BC%A0)，已经上传
> > > 
> > > 
> > > 测试 转发53端口到adguard 模式是无效的。此模式下 无法过滤广告，面板里查询日志为0
> > 
> > 
> > ![a724359f-d81e-4126-9245-8ff086330283](https://private-user-images.githubusercontent.com/20589467/393231869-ba4fbeb7-3ab8-4434-b731-69c6cef7a54a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM1MzIxNzMsIm5iZiI6MTczMzUzMTg3MywicGF0aCI6Ii8yMDU4OTQ2Ny8zOTMyMzE4NjktYmE0ZmJlYjctM2FiOC00NDM0LWI3MzEtNjljNmNlZjdhNTRhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjA3VDAwMzc1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBlNDY3MzMwNzk5ZGIwMjFlMTMzYTFmYTY1MGRkOGU0MTI5M2Q5YjRlZTkwNjBmZWNmZTBlYzc5Yjg1ZTFkNmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.lpMvgLFLh_k2wOquo_CzsSxQ6ZIPBiJQ2ZhtA6eigB8) 很奇怪。。我也是同一个版本，这边测试是可以的，你好固件确定是纯fw4么，没有混合fw3在里面吧
> 
> ============================================================================ 应该是纯fw4的。特地重置了一边。在系统防火墙里看状态 没提示有iptables规则（如果有的话会提示）。固件是25.0.4 ，已安装包里也只有firewall4。一旦改成adguard当上游，立刻就有过滤了，但是此模式下因为无法看客户端ip，所以不是我喜欢的模式 ![捕获3](https://private-user-images.githubusercontent.com/39509734/393453706-3561d8ca-9ea6-40e9-a200-cf09083c723a.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM1ODczMjQsIm5iZiI6MTczMzU4NzAyNCwicGF0aCI6Ii8zOTUwOTczNC8zOTM0NTM3MDYtMzU2MWQ4Y2EtOWVhNi00MGU5LWEyMDAtY2YwOTA4M2M3MjNhLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjA3VDE1NTcwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI0YWMyMjZjNWM2ODBkYjk1YmJiMjhkOTdmMzVlYWU1Yjg1NzM1YTI2ZmYyNThiZjQ0YTM4NDY3ZGI2YjlkOTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.UnigILnOEOj0lmeh0lrJM3oFT2Gltmeymysqc28Lc1s) ![捕获2](https://private-user-images.githubusercontent.com/39509734/393453721-3258c17b-f3cc-4efa-b57e-6fcae5a2d239.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM1ODczMjQsIm5iZiI6MTczMzU4NzAyNCwicGF0aCI6Ii8zOTUwOTczNC8zOTM0NTM3MjEtMzI1OGMxN2ItZjNjYy00ZWZhLWI1N2UtNmZjYWU1YTJkMjM5LlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjA3VDE1NTcwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI0ZjUxZjU1ZWE5NDA4NWM3MzAxYTE0NjkyZjYxODhjMDcxNTk0YjllNzE5ZjVmZmE1MWIyMGFiZDlmYjBkZWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.AT2Rr6WzfCS2K5pTygsAGskhbdE1P4aulKbaVXH4Fj8) ![捕获](https://private-user-images.githubusercontent.com/39509734/393453739-f671cf7a-1743-483f-9cf0-6f1e42a7b6ee.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM1ODczMjQsIm5iZiI6MTczMzU4NzAyNCwicGF0aCI6Ii8zOTUwOTczNC8zOTM0NTM3MzktZjY3MWNmN2EtMTc0My00ODNmLTljZjAtNmYxZTQyYTdiNmVlLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjA3VDE1NTcwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYzMDY5MTJiODg0MzlkYTljNjU4ZDM1NDBjYjMyMzc5NjFkZWNiMzVhZTgzMzNmYzUxNzcxNDIzZWY3ZTEwMTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.UfXFy-HcRvabz55f9ghq077OOt96tEsGT8Eqf9eht7c)

我这边测试了几次都没法重现。。点击强制更新核心，看看会不会出现什么报错

MIAOBUSI, 2024-12-07:
> 我这边测试了几次都没法重现。。点击强制更新核心，看看会不会出现什么报错

[image]
我看你的截图这最上面有个黄色提示，这个正常来说就是提示存在fw3和fw4混合规则

GoogleBeEvil, 2024-12-08:
> ![image](https://private-user-images.githubusercontent.com/20589467/393531036-f0473867-f646-4593-8fb9-8765901eb2ac.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

那个是root登录警告
[image]

MIAOBUSI, 2024-12-08:
> > > > > > > > >
> > 
> > 
> > ![image](https://private-user-images.githubusercontent.com/20589467/393531036-f0473867-f646-4593-8fb9-8765901eb2ac.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM2Mjc0MjgsIm5iZiI6MTczMzYyNzEyOCwicGF0aCI6Ii8yMDU4OTQ2Ny8zOTM1MzEwMzYtZjA0NzM4NjctZjY0Ni00NTkzLThmYjktODc2NTkwMWViMmFjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjA4VDAzMDUyOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk1NDg4MDA4OGIxMzA5NDM1MzVmZDcwNWU2OWE5MzhiMzg1MGJlNGJmOWIxODJlZTc2YWFmZDE5OTg0MTlhNjMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.lnDXHyUjL_ZSTG0ADDR8ics-F3bCEvnqnVIVbOorSiE) 我看你的截图这最上面有个黄色提示，这个正常来说就是提示存在fw3和fw4混合规则
> 
> # 那个是root登录警告
> ![2024-12-08_110703](https://private-user-images.githubusercontent.com/39509734/393560390-e10a4dc1-0dbc-436e-a75b-e3fea97cca0c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM2Mzc5MDksIm5iZiI6MTczMzYzNzYwOSwicGF0aCI6Ii8zOTUwOTczNC8zOTM1NjAzOTAtZTEwYTRkYzEtMGRiYy00MzZlLWE3NWItZTNmZWE5N2NjYTBjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjA4VDA2MDAwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNlZjA5NzNmY2U5MmNkN2E4NzI3YzhmOGU0NjMwZjBiYjU5MDViYTI2NGUzNjlhMGNiNjc5ZjE2OTkyODRmMTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.k84eJD_7yZBW1NTztbp58-xN-7nOTvi7OlWs8_rH0k0)
> 强制更新日志似乎没啥问题
> 
> ```
> - 96 9749k   96 9390k    0     0  86686      0  0:01:55  0:01:50  0:00:05  125k
> -  97 9749k   97 9521k    0     0  87099      0  0:01:54  0:01:51  0:00:03  124k
> -  99 9749k   99 9694k    0     0  87899      0  0:01:53  0:01:52  0:00:01  130k
> - 100 9749k  100 9749k    0     0  87971      0  0:01:53  0:01:53 --:--:--  128k
> - download success start copy
> - start copy
> - AdGuardHome service disabled
> - Command failed: ubus call service delete { "name": "AdGuardHome", "instance": "nobackup" } (Not found)
> - workdir is a f2fs filesystem
> - AdGuardHome service enabled
> - luci enable switch=1
> - Succeeded in updating core.
> - Local version: v0.107.54, cloud version: v0.107.54.
> ```
> 
> ```
> - 2024/12/08 11:20:49.302394 6014#701 [debug] auth: removed session from DB 
> - 2024/12/08 11:20:49.302587 6014#701 [debug] auth: raddr 192.168.111.102:50162: invalid cookie value: "agh_session=2137299036b530e04726a6e2db5abc62"
> - 2024/12/08 11:20:49.302707 6014#701 [debug] auth: raddr 192.168.111.102:50162: redirected to login page
> - 2024/12/08 11:20:49.328741 6014#701 [debug] auth: raddr 192.168.111.102:50162: invalid cookie value: "agh_session=2137299036b530e04726a6e2db5abc62"
> - 2024/12/08 11:20:55.111996 6014#701 [debug] started POST 192.168.111.1:3000 /control/login
> - 2024/12/08 11:20:55.377135 6014#701 [debug] auth: created session c9bd1741573a3204ac44c4381326aa04: expire=1733629255
> - 2024/12/08 11:20:55.377302 6014#701 [info] auth: user "root" successfully logged in from ip 192.168.111.102
> - 2024/12/08 11:20:55.377477 6014#701 [debug] finished POST 192.168.111.1:3000 /control/login in 265.536214ms
> - 2024/12/08 11:20:56.331534 6014#701 [debug] started GET 192.168.111.1:3000 /control/status
> - 2024/12/08 11:20:56.333207 6014#701 [debug] finished GET 192.168.111.1:3000 /control/status in 1.662865ms
> - 2024/12/08 11:20:56.949802 6014#701 [debug] started GET 192.168.111.1:3000 /control/access/list
> - 2024/12/08 11:20:56.950093 6014#701 [debug] finished GET 192.168.111.1:3000 /control/access/list in 327.784µs
> - 2024/12/08 11:20:56.965084 6014#701 [debug] started GET 192.168.111.1:3000 /control/stats
> - 2024/12/08 11:20:56.965581 6014#701 [debug] stats: loading unit id=481540
> - 2024/12/08 11:20:56.966096 6014#701 [debug] stats: loading unit id=481541
> - 2024/12/08 11:20:56.966600 6014#701 [debug] stats: loading unit id=481542
> - 2024/12/08 11:20:56.966605 6014#718 [debug] started GET 192.168.111.1:3000 /control/stats/config
> - 2024/12/08 11:20:56.966952 6014#718 [debug] finished GET 192.168.111.1:3000 /control/stats/config in 379.516µs
> - 2024/12/08 11:20:56.967062 6014#701 [debug] stats: loading unit id=481543
> - 2024/12/08 11:20:56.967498 6014#701 [debug] stats: loading unit id=481544
> - 2024/12/08 11:20:56.967932 6014#701 [debug] stats: loading unit id=481545
> - 2024/12/08 11:20:56.968333 6014#701 [debug] stats: loading unit id=481546
> - 2024/12/08 11:20:56.968800 6014#701 [debug] stats: loading unit id=481547
> - 2024/12/08 11:20:56.969244 6014#701 [debug] stats: loading unit id=481548
> - 2024/12/08 11:20:56.969623 6014#701 [debug] stats: loading unit id=481549
> - 2024/12/08 11:20:56.970068 6014#701 [debug] stats: loading unit id=481550
> - 2024/12/08 11:20:56.970350 6014#756 [debug] dnsforward: dialing "static.adtidy.org:443" for network "tcp"
> - 2024/12/08 11:20:56.970560 6014#701 [debug] stats: loading unit id=481551
> - 2024/12/08 11:20:56.971143 6014#701 [debug] stats: loading unit id=481552
> - 2024/12/08 11:20:56.971753 6014#701 [debug] stats: loading unit id=481553
> - 2024/12/08 11:20:56.972200 6014#701 [debug] stats: loading unit id=481554
> - 2024/12/08 11:20:56.972661 6014#701 [debug] stats: loading unit id=481555
[... cut ...] G09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjA4VDA2MDAwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE1ZjQ3MWU5NjBiM2YwNzIxZmNkNTBkNDVlYTg5NjJjZTU3MGRjZGRmMjc1ZDFlMDhjNjYzZWJhZDdlYjRkYWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.r6UJ_-7zyZ7X79ARQAm5PmIl8gXzz0KHfuSPt4g8ngA)
[image]

那我知道是什么问题了，这个文件确实权限，你要用winscp登录openwrt，然后进入到/etc/init.d文件夹，找到AdGuardHome文件，将权限改成下图所示，然后再试试。
[image]
[image]

GoogleBeEvil, 2024-12-08:
> 那我知道是什么问题了，这个文件确实权限，你要用winscp登录openwrt，然后进入到/etc/init.d文件夹，找到AdGuardHome文件，将权限改成下图所示，然后再试试。 [...]

实际上  我第一次安装完就赋权775了 ，但是没效果

MIAOBUSI, 2024-12-08:
> 实际上 我第一次安装完就赋权775了 ，但是没效果

[image]
查看一下这个位置是否正确添加这些子项

GoogleBeEvil, 2024-12-10:
> > > > > > > > > >

> 查看一下这个位置是否正确添加这些子项

[image]

MIAOBUSI, 2024-12-10:
> > > > > > > > > > >
> 
> > 查看一下这个位置是否正确添加这些子项
> 
> ![捕获](https://private-user-images.githubusercontent.com/39509734/394242673-8702eb6b-6a76-4a63-8a32-2cc0a9db3027.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM4NDUxMzksIm5iZiI6MTczMzg0NDgzOSwicGF0aCI6Ii8zOTUwOTczNC8zOTQyNDI2NzMtODcwMmViNmItNmE3Ni00YTYzLThhMzItMmNjMGE5ZGIzMDI3LlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjEwVDE1MzM1OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI3NTYwZDkyNGNhYzhjOTViYzA4NzMxYTdjZTExOTYyYzE1ZDhmODYwM2Y4MDk0NTg3MDI1YTMyMTdiZjM2NGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.oCfH2cXK464rD6B4JEAwQbaQ8yETpcwpD7jPcn96EL4)

[AdGuardHome.zip](https://github.com/user-attachments/files/18082010/AdGuardHome.zip)
你这个是旁路由么？将这个文件替换之/etc/init.d下，然后再试试吧。

GoogleBeEvil, 2024-12-11:
> [AdGuardHome.zip](https://github.com/user-attachments/files/18082010/AdGuardHome.zip) 你这个是旁路由么？将这个文件替换之/etc/init.d下，然后再试试吧。

无效的，我这个路由是二级路由，用无线中继上级路由

MIAOBUSI, 2024-12-11:
> 无效的，我这个路由是二级路由，用无线中继上级路由

二级路由那就不知道了，只有一级路由的话测试是没问题

GoogleBeEvil, 2024-12-12:
应该是规则有问题  这个方法在我是生效的。他的规则写法和你的似乎有些区别

> ```shell
> 添加 转发到adguardhome 1053端口
> nft add table inet adguardhome
> nft add chain inet adguardhome prerouting "{ type nat hook prerouting priority -110; policy accept; }"
> nft add rule inet adguardhome [...]

MIAOBUSI, 2024-12-29:
> 应该是规则有问题 下面这个方法在我是生效的。他的规则写法和你的似乎有些区别，先生成了表和链，再把规则添加进链，看你的代码似乎只是把iptable转成了nft，没有添加表、链这一步
> 
> > ```shell
> > 添加 转发到adguardhome 1053端口
> > nft add table inet adguardhome
> > nft add chain inet adguardhome prerouting "{ type nat hook prerouting priority -110; policy accept; }"
> > nft add rule inet adguardhome prerouting "meta nfproto { ipv4, ipv6 } udp dport 53 counter redirect to :1053 comment \"ADGUARDHOME HIJACK\""
> > 
> > # 清理
> > ! nft --check list table inet adguardhome > "/dev/null" 2>&1 || nft delete table inet adguardhome
> > ```
> 
> 另外：最新版的openwrt 23.05.5 安装已经无效了，连菜单都不会出现

[image]
[image]
我今天抽空重新安装了系统，并使用的是immowrt 24.10版本依然没出现问题

liyiranlab, 2024-12-29:
大佬，nb，之前用的luci全部都无法联网，明天试试你的这个。

MIAOBUSI, 2024-12-30:
> 大佬，nb，之前用的luci全部都无法联网，明天试试你的这个。

昨天重装发现AdGuardHome存在权限问题，首次安装需要自己手动用winscp登录openwrt，然后进入到/etc/init.d文件夹，找到AdGuardHome文件，将权限改成0777

liyiranlab, 2024-12-30:
更新到主线最新的版本，无法联网，提示Failed to check latest version, please try again later.
我用的AX6是nss版本，不知道是什么问题。后面再看看别的版本。

liyiranlab, 2024-12-30:
24.10也是一样，我试试不用nss版本，看看是否正常。

GoogleBeEvil, 2024-12-30:
看来我得弄个虚拟机镜像  来测试了

MIAOBUSI, 2024-12-30:
> 更新到主线最新的版本，无法联网，提示Failed to check latest version, please try again later. 我用的AX6是nss版本，不知道是什么问题。后面再看看别的版本。

Failed to check latest version, please try again later.这个是你无法访问github把，路由的科学上网是否正常呢，因为下载镜像是需要通过github下载

MIAOBUSI, 2024-12-30:
> 看来我得弄个虚拟机镜像 来测试了

换个环境测试下试试吧，目前我发现的就只是首次安装后/etc/init.d/AdGuardHome文件存在权限问题，手动重新修改文件权限就没问题了

liyiranlab, 2024-12-30:
> > 更新到主线最新的版本，无法联网，提示Failed to check latest version, please try again later. 我用的AX6是nss版本，不知道是什么问题。后面再看看别的版本。
> 
> Failed to check latest version, please try again later.这个是你无法访问github把，路由的科学上网是否正常呢，因为下载镜像是需要通过github下载

Failed to check latest version, please try again later.
/usr/bin/curl 
imm天灵主线360T7，还是无法访问，用带bin的adg可以使用，但是无法更新，我上级路由器是代理了的，浏览器也可以正常打开下载目录。更多信息我不知道去哪看？

liyiranlab, 2024-12-30:
我感觉是主线改了一些东西，之前wifi配置无法保存，hp无法保存，前几天才修复的。

liyiranlab, 2024-12-30:
另外，360T7系统日志提示：Mon Dec 30 12:51:34 2024 daemon.err uhttpd[2159]: grep: /etc/AdGuardHome.yaml: No such file or directory

MIAOBUSI, 2024-12-30:
> Failed to check latest version, please try again later. /usr/bin/curl imm天灵主线360T7，还是无法访问，用带bin的adg可以使用，但是无法更新，我上级路由器是代理了的，浏览器也可以正常打开下载目录。更多信息我不知道去哪看？

你的是属于旁路由还是二级路由，如果属于旁路由是要指定DNS，二级路由不应该会出现这种问题

MIAOBUSI, 2024-12-30:
> 另外，360T7系统日志提示：Mon Dec 30 12:51:34 2024 daemon.err uhttpd[2159]: grep: /etc/AdGuardHome.yaml: No such file or directory

这个是提示 /etc/AdGuardHome.yaml这个目录下配置文件不存在

liyiranlab, 2024-12-30:
二级路由和旁路由都有，有nss的二级路由，测试用的，T7是旁路由，接成ap使用中，旁路由自身关闭ipv6防止污染，passwall测试旁路由自身ipv4是通的，上级主路由代理，t7旁路由翻墙也正常。

liyiranlab, 2024-12-30:
/usr/bin/curl
Local version: v0.107.53., cloud version: v0.107.55.
Updating core...
/usr/share/AdGuardHome/update_core.sh: line 125: opkg: not found
error not support  if you can use offical release please issue a bug
主路由AX6nss，用的是 https://github.com/kenzok8/openwrt-packages 源码。

liyiranlab, 2024-12-30:
启动项 没有adg

liyiranlab, 2024-12-30:
用上面带bin的版本，openwrt主线能开启但是无法正常使用。

MIAOBUSI, 2024-12-30:
> 用上面带bin的版本，openwrt主线能开启但是无法正常使用。

我那个版本启动项没有adg么

MIAOBUSI, 2024-12-30:
> 用上面带bin的版本，openwrt主线能开启但是无法正常使用。

我的版本启动项没有adg是正常的，因为我的核心是独立的，它可以在luci上面进行安装更新的

MIAOBUSI, 2024-12-30:
> 你的好像启动项好像也没有，带bin的版本不能更新，可以启动，可以打开界面，但是好像工作不正常，我设置53端口，但是没捕捉到几个客户端，实际数据不对。

我这个版本是没有启动项的，因为是不带核心的版本，核心是独立的，它只是一个用于启动和调用adg的luci

liyiranlab, 2024-12-31:
不能吧，我写了个ipv6监控的脚本，放init.d里面都有启动项。

MIAOBUSI, 2024-12-31:
> 不能吧，我写了个ipv6监控的脚本，放init.d里面都有启动项。

[image]
这是我自己下面运行脚本 依然是没有，但是是正常使用的

liyiranlab, 2024-12-31:
可能有错误，导致没有正常加载，init.d有文件就应该有。

ywmail, 2025-06-18:
使用最后一个选项解决了：使用53端口替换dnsmasq