rufengsuixing/luci-app-adguardhome #147: 添加gfw后启动不了，取消后成功启动

NoNameCanUse9, 2024-03-02:
### Issue Details
如题  添加gfw后启动不了，取消后成功启动

* **Version of AdGuard Home luci:**
  * v1.0-7
* **Version of AdGuard Home server:**
  * 
  * v0.107.44
* **Operating system and version:**
  * 

### Screenshots

<details><summary>Screenshot:</summary>

[image]
[image]

</details>

log信息
2024/03/02 18:21:18.051652 26164#1 [error] parsing configuration file: parsing config file for upgrade: yaml: line 15: did not find expected key
2024/03/02 18:21:18.050078 26164#1 [debug] reading config file: /etc/AdGuardHome.yaml
2024/03/02 18:21:18.049520 26164#1 [debug] current working directory is /usr/bin/AdGuardHome
2024/03/02 18:21:18.048947 26164#1 [info] AdGuard Home, version v0.107.44

### Additional Information

yaml文件
dns:
  bind_hosts:
    - 0.0.0.0
  port: 533
  anonymize_client_ip: false
  ratelimit: 20
  ratelimit_subnet_len_ipv4: 24
  ratelimit_subnet_len_ipv6: 56
  ratelimit_whitelist: []
  refuse_any: true
  upstream_dns:
  - '[/programaddstart/]#'
  - '[/programaddend/]#'
    - https://doh.pub/dns-query
    - https://dns.alidns.com/dns-query
  upstream_dns_file: ""
  bootstrap_dns:
    - 202.96.134.33
  fallback_dns: []
  upstream_mode: parallel
  fastest_timeout: 1s
  allowed_clients: []
  disallowed_clients: []
  blocked_hosts:
    - version.bind
    - id.server
    - hostname.bind
  trusted_proxies:
    - 127.0.0.0/8
    - ::1/128
  cache_size: 4194304
  cache_ttl_min: 0
  cache_ttl_max: 0
  cache_optimistic: false
  bogus_nxdomain: []
  aaaa_disabled: false
  enable_dnssec: false
  edns_client_subnet:
    custom_ip: ""
    enabled: false
    use_custom: false
  max_goroutines: 300
  handle_ddr: true
  ipset: []
  ipset_file: ""
  bootstrap_prefer_ipv6: false
  upstream_timeout: 10s
  private_networks: []
  use_private_ptr_resolvers: true
  local_ptr_upstreams: []
  use_dns64: false
  dns64_prefixes: []
  serve_http3: false
  use_http3_upstreams: false
  serve_plain_dns: true

hbcraft, 2024-07-14:
有解决方法了吗？

NoNameCanUse9, 2024-07-21:
> 有解决方法了吗？

没有，已经不用了，

ggtony233, 2024-09-24:
脚本的问题，yaml文件缩进不正确就会出现无法启动的情况，要么让开发者自己修改脚本搞定缩进的问题，要么把gfw的上游列表放到upstream_dns_file这个设置项里面
@rufengsuixing