rufengsuixing/luci-app-adguardhome #154: 在服务正常运行之后，无法登录，日志提示 invalid cookie value:

leejoa, 2024-12-02:
### 我的日志提示记录，是否配置异常
2024/12/02 16:47:35.722572 28373#91 [debug] auth: raddr 192.168.88.88:60700: invalid cookie value: "agh_session=cec4747720e6b8200029ee696a473b39"
2024/12/02 16:47:35.714965 28373#91 [debug] auth: raddr 192.168.88.88:60700: redirected to login page
2024/12/02 16:47:35.714907 28373#91 [debug] auth: raddr 192.168.88.88:60700: invalid cookie value: "agh_session=cec4747720e6b8200029ee696a473b39"
2024/12/02 16:47:35.714800 28373#91 [debug] auth: removed session from DB
2024/12/02 16:47:35.515310 28373#91 [debug] finished POST 192.168.88.2:3012 /control/login in 332.734389ms
2024/12/02 16:47:35.515181 28373#91 [info] auth: user "root" successfully logged in from ip 192.168.88.88
2024/12/02 16:47:35.515077 28373#91 [debug] auth: created session cec4747720e6b8200029ee696a473b39: expire=1733129255
2024/12/02 16:47:35.182606 28373#91 [debug] started POST 192.168.88.2:3012 /control/login
2024/12/02 16:47:29.640743 28373#91 [debug] auth: raddr 192.168.88.88:60700: redirected to login page
2024/12/02 16:45:50.252380 28373#76 [debug] auth: raddr 192.168.199.199:50555: invalid cookie value: "agh_session=0b89b33267414dd0a2df3e1328d49b0f"
2024/12/02 16:45:50.236738 28373#76 [debug] auth: raddr 192.168.199.199:50555: redirected to login page
2024/12/02 16:45:50.236685 28373#76 [debug] auth: raddr 192.168.199.199:50555: invalid cookie value: "agh_session=0b89b33267414dd0a2df3e1328d49b0f"
2024/12/02 16:45:50.236595 28373#76 [debug] auth: removed session from DB
2024/12/02 16:45:49.992741 28373#76 [debug] finished POST 192.168.88.2:3012 /control/login in 385.317584ms
2024/12/02 16:45:49.992656 28373#76 [info] auth: user "root" successfully logged in from ip 192.168.199.199
2024/12/02 16:45:49.992571 28373#76 [debug] auth: created session 0b89b33267414dd0a2df3e1328d49b0f: expire=1733129149
2024/12/02 16:45:49.607453 28373#76 [debug] started POST 192.168.88.2:3012 /control/login
2024/12/02 16:45:08.579423 28373#76 [debug] auth: raddr 192.168.199.199:50555: invalid cookie value: "agh_session=ca5a89b1c704ce5aa8e75cf2a1e1e4d2"
2024/12/02 16:45:08.563538 28373#76 [debug] auth: raddr 192.168.199.199:50555: redirected to login page
2024/12/02 16:45:08.563470 28373#76 [debug] auth: raddr 192.168.199.199:50555: invalid cookie value: "agh_session=ca5a89b1c704ce5aa8e75cf2a1e1e4d2"
2024/12/02 16:45:08.563386 28373#76 [debug] auth: removed session from DB
2024/12/02 16:45:08.352813 28373#76 [debug] finished POST 192.168.88.2:3012 /control/login in 316.203415ms
2024/12/02 16:45:08.352663 28373#76 [info] auth: user "root" successfully logged in from ip 192.168.199.199
2024/12/02 16:45:08.352557 28373#76 [debug] auth: created session ca5a89b1c704ce5aa8e75cf2a1e1e4d2: expire=1733129108
2024/12/02 16:45:08.036618 28373#76 [debug] started POST 192.168.88.2:3012 /control/login
2024/12/02 16:45:00.610175 28373#76 [debug] auth: raddr 192.168.199.199:50555: invalid cookie value: "agh_session=f23242559bf41520b398e5d02be01970"
2024/12/02 16:45:00.597302 28373#76 [debug] auth: raddr 192.168.199.199:50555: redirected to login page
2024/12/02 16:45:00.597233 28373#76 [debug] auth: raddr 192.168.199.199:50555: invalid cookie value: "agh_session=f23242559bf41520b398e5d02be01970"
2024/12/02 16:43:27.632201 28373#65 [debug] auth: raddr 192.168.199.199:50509: invalid cookie value: "agh_session=f23242559bf41520b398e5d02be01970"
2024/12/02 16:43:27.603769 28373#65 [debug] auth: raddr 192.168.199.199:50509: redirected to login page
2024/12/02 16:43:27.603711 28373#65 [debug] auth: raddr 192.168.199.199:50509: invalid cookie value: "agh_session=f23242559bf41520b398e5d02be01970"
2024/12/02 16:43:27.603595 28373#65 [debug] auth: removed session from DB
2024/12/02 16:43:27.248700 28373#65 [debug] finished POST 192.168.88.2:3012 /control/login in 322.315909ms
2024/12/02 16:43:27.248618 28373#65 [info] auth: user "root" successfully logged in from ip 192.168.199.199
2024/12/02 16:43:27.248535 28373#65 [debug] auth: created session f23242559bf41520b398e5d02be01970: expire=1733129007
2024/12/02 16:43:26.926414 28373#65 [debug] started POST 192.168.88.2:3012 /control/login
2024/12/02 16:43:19.664108 28373#65 [debug] auth: raddr 192.168.199.199:50509: invalid cookie value: "agh_session=778caefede8785be69af7103bc37e7b5"
2024/12/02 16:41:51.829844 28373#56 [debug] auth: raddr 192.168.199.199:50439: invalid cookie value: "agh_session=778caefede8785be69af7103bc37e7b5"
2024/12/02 16:41:51.814576 28373#56 [debug] auth: raddr 192.168.199.199:50439: redirected to login page
2024/12/02 16:41:51.814517 28373#56 [debug] auth: raddr 192.168.199.199:50439: invalid cookie value: "agh_session=778caefede8785be69af7103bc37e7b5"
2024/12/02 16:41:51.814392 28373#56 [debug] auth: removed session from DB
2024/12/02 16:41:51.603814 28373#56 [debug] finished POST 192.168.88.2:3012 /control/login in 304.747317ms
2024/12/02 16:41:51.603730 28373#56 [info] auth: user "root" successfully logged in from ip 192.168.199.199
2024/12/02 16:41:51.603642 28373#56 [debug] auth: created session 778caefede8785be69af7103bc37e7b5: expire=1733128911
2024/12/02 16:41:51.299097 28373#56 [debug] started POST 192.168.88.2:3012 /control/login
2024/12/02 16:41:46.695563 28373#56 [debug] finished POST 192.168.88.2:3012 /control/login in 270.887µs
2024/12/02 16:41:46.695479 28373#56 [error] POST 192.168.88.2:3012 /control/login: from ip 192.168.199.199: invalid username or password
2024/12/02 16:41:46.695308 28373#56 [debug] started POST 192.168.88.2:3012 /control/login

leejoa, 2024-12-02:
> ### 我的日志提示记录，是否配置异常
> 2024/12/02 16:47:35.722572 28373#91 [debug] auth: raddr 192.168.88.88:60700: invalid cookie value: "agh_session=cec4747720e6b8200029ee696a473b39" 2024/12/02 16:47:35.714965 [...]

已经尝试更换过登录的PC。依旧报错。

GoogleBeEvil, 2024-12-20:
session_ttl: 720h