rufengsuixing/luci-app-adguardhome #155: JCG Q20路由器，显示core error 没有配置文件no core没有核心

ningboyun, 2025-04-26:
### 描述

显示core error 没有配置文件no core没有核心

点击“更新更新核心版本”，最后提示：no download success
然后然后软件包中下载“adguardhome”并安装成功，版本是：0.107.57-r1。
并且进入192.168.2.1:3000 能进入dguardhome管理页面

### 更新更新核心版本日志
 Local version: ., cloud version: v0.107.61
v0.107.61
v0.107.61
v0.107.61
v0.107.61
v0.107.61
v0.107.61
v0.107.61
v5.
v0.107.61
v5.
v6.
v0.107.61
v6.
v7.
v0.107.61
v7.
v0.107.61
v0.107.61
v0.107.61
v0.107.61
v5.
v0.107.61
v5.
v6.
v0.107.61
v6.
v7.
v0.107.61
v7.
v0.107.61
v0.107.61
v0.107.61
v0.107.61
v0.107.61
v64.
v0.107.61
v64.
v0.107.61
v0.107.61
v0.107.61
v0.107.61
v0.107.61
v0.107.61
v0.107.61
v0.107.61
v0.107.61
v0.107.61
v0.107.61.
Updating core...
start download
/usr/share/AdGuardHome/update_core.sh: line 168: wget-ssl: not found
download failed try another download
/usr/share/AdGuardHome/update_core.sh: line 168: wget-ssl: not found
download failed try another download
no download success

### 截图

[image]

[image]

ningboyun, 2025-04-27:
已解决，亲测有效

日志报的错误：
start download
/usr/share/AdGuardHome/update_core.sh: line 168: wget-ssl: not found
download failed try another download
/usr/share/AdGuardHome/update_core.sh: line 168: wget-ssl: not found
download failed try another download
no download success

意思是文件/usr/share/AdGuardHome/update_core.sh的第168行，wget-ssl命令无法运行。

解决方法：
1. 打开文件：/usr/share/AdGuardHome/update_core.sh定位到168行，把 `wget-ssl` 改成 `wget` 
2. 在【系统】-【软件包】中搜索wget，安装wget-ssl程序包

[image]

4. 再次运行更新核心

[image]

yeeyrr, 2025-09-22:
你这个的版本是0.102.0，我好奇怎么使用最新版